Neotron-Compute/Neotron-BMC #19: Run on a Nucleo-F031K6

jonathanpallant, 2022-01-27:
We should have a feature for running on a Nucleo board, seeing as we can't make Neotron PCBs until a bunch more parts come into stock. It also rules out issues with my bad PCB design.

The main difference is the Nucleo has a crystal fitted to PF0/PF1 so the button inputs might not work. But if the Crystal oscillator isn't enabled, maybe each end of the crystal won't mind being grounded with a button input. Or you could live without buttons. Or live without something else instead.

thejpster, 2022-02-02:
OK, we're doing with a Nucleo-64 carrying an STM32F401 because @Dajamante has one to hand.

Dajamante, 2022-02-02:
Shall we close this one then?

thejpster, 2022-02-02:
It doesn't run on an F401 yet ;)

closed 2022-03-03 by jonathanpallant